dekuNukem/daytripper #8: Interested in making these my control my lights

dude2k5, 2019-09-12:
I've always needed something like this, there was no tech or sensor or signal that would detect when someone walked into rooms. Can you keep these plugged in so they are on at all times? Im going to try one in my bedroom, when i walk in, hopefully i can run a script that will turn on a wemo light. Then turn it off when i walk out. If it works, this would be AMAZING. Very very cool project.

nh2, 2019-09-14:
Interested in that too. Thanks for answering the "keep plugged in" question, that was also my first one.

Second, can you use (and purchase) multiple TX with a single RX, so that you can use one RX "base station" to handle multiple doors?

dekuNukem, 2019-09-14:
I touched upon this topic in [advanced usage guide](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#using-multiple-modules) .

Basically, daytripper wasn't really designed with multiple TX or RX in mind, so I didn't to much testing. However Single RX multiple TX does seem to work, and RX will respond with ANY TX triggers.

nh2, 2019-09-15:
Thanks! When one of multipe TX triggers, can the RX figure out which one it was?

dekuNukem, 2019-09-15:
At the moment no. RX will do the same thing if any of the multiple TX triggers.

However, each TX does include a unique ID in its [on-air-packet](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#on-air-packet-format). So if you're willing to do a bit programming, it is completely possible to assign different actions to different TXs. 

TX also prints out debug information in its debug serial harder, so that's another way to tell.

nh2, 2019-09-15:
> However, each TX does include a unique ID in its [on-air-packet](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#on-air-packet-format). So if you're willing to do a bit programming

That's great, that'll do.

I've ordered one just now to try it out.

dekuNukem, 2019-09-15:
Thanks, the relevant code is [around here](https://github.com/dekuNukem/daytripper/blob/3fa12687489bfb83d95f485eb91f7c48078b31e9/firmware/rx/Src/main.c#L224) in the RX firmware. `received_data[0]` is the unique ID, you can check that and do your stuff.

If you're unfamiliar with STM32 dev, I have wrote [a tutorial](https://github.com/dekuNukem/STM32_tutorials) about it too.

nh2, 2019-09-15:
Good stuff, very appreaciated!

That's how modern hardware should work: Easy to order, easy to modify for your needs. :+1:

Slomo17, 2019-10-30:
> At the moment no. RX will do the same thing if any of the multiple TX triggers.
> 
> However, each TX does include a unique ID in its [on-air-packet](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#on-air-packet-format). So if you're willing to do a bit programming, it is completely possible to assign different actions to different TXs.
> 
> RX also prints out debug information in its debug serial harder, so that's another way to tell.

I have a simmilar project, which may be possible with the daytripper.
For years i wanted to build something, that pauses the movie i am watching when i leave the room, and resumes it when i am back (i use kodi as a media center on a raspberry pi). The main problem here is if there are more than just me in the room and another person leaves the room it would resume the movie. So i would need something that can detect the direction of the person crossing the laser, for which i would need two lasers. With the ID you mention it may be possible for me to use two TX und figure out which one got activated first to determin the direction