dekuNukem/daytripper #11: About firmware

oniyakun, 2019-09-17:
I'm ordered some STM32f6p6 chips and try to writing the firmware `rx.hex` and `tx.hex` into it, but the firmware is much much larger than the chip capacity then I cannot write the firmware. plz help me, what should I do?

dekuNukem, 2019-09-17:
The chip that both TX and RX uses is STM32F042F6P6, so first make sure the chip is correct. 

Then if you have the boards assembled, you can upload their firmware through USB DFU, which is described here:

https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#usb-firmware-updates

oniyakun, 2019-09-17:
Oh so u means I could upload the firmware by using the USB DFU software even the chip is empty? I'm worry about that cloud I using USB DFU to upload the firmware into an empty chip, huge thanx!

oniyakun, 2019-09-17:
And there is any changed component in the PCB files since the last time u upload?

dekuNukem, 2019-09-17:
I havent made any changes to the components on the PCB. And yes USB DFU works even when the chip is empty :)

oniyakun, 2019-09-17:
Got it, thx a lot!

closed 2019-09-17 by oniyakun