dekuNukem/daytripper #15: TX unit not working

cgoodman1976, 2019-11-07:
Hello, no matter what I try, the TX unit appears to not function. I press the button expecting a message to be sent to the RX unit and it does not register. When I press the button on the RX , I get a prompt back like this: 

transmitter ID: 0x0
baseline: 0mm
latest trigger: 0mm
input voltage: 0mV
power-on time: 0 minutes

Please help.

dekuNukem, 2019-11-07:
Hi:

I'm really sorry about this issue, it was discovered recently and rather caught me off guard.

I wrote a short article, including how to fix it. Please take a look here:

https://github.com/dekuNukem/daytripper/blob/master/mysterious_glitch.md

If you do manage to fix it, let me know your Tindie order number and I'll give you a partial refund for the trouble. And if that didn't work, I'll mail you a new one for free.

cgoodman1976, 2019-11-08:
howdy! I did the instructions as you suggested and all is well. The unit is
working as expected.

Thanks!

-cg

dekuNukem, 2019-11-08:
Great! I'll ask Tindie to give you a partial refund for the trouble. You should get it within 48 hours.

closed 2019-11-08 by dekuNukem

varenc, 2020-03-27:
@dekuNukem I'd humbly recommend you make the glitch doc a bit easier to find. Maybe a link on the readme or in quick start guide?

I spent some time looking through the advanced doc and recent issues before I found it.  I suspect #18 has the same problem but it doesn't look like they found the mysterious glitch doc.  Thanks for documenting it so thoroughly!  Super interesting how such a subtle thing can cause issues.

ringnex1, 2020-04-08:
hey varenc did your rx start working again because I had that problem but my tx worked fine if you did can help me with it

varenc, 2020-04-09:
so I think my RX was actually working fine all along. How do you know your TX is working correctly?  Are you sure you don't have the battery contact problem that's [documented here](https://github.com/dekuNukem/daytripper/blob/master/mysterious_glitch.md)?

The RX's debug button seems to return all zeros if it's never paired/seen the TX so if you're seeing all zeros that doesn't necessarily mean the issue is with the RX.  Could be that TX is not transmitting correctly (what happened with me), or could mean that the RX for some reason isn't able to correctly receive anything, but based on my limited experience my guess is it's an TX transmission issue.

I open PR #19 to make a tiny note of this in the advanced usage guide.