dekuNukem/daytripper #16: Disable Blue LED

PCBuildr, 2019-11-27:
Is there a way to disable the blue LED light on the TX when the sensor is triggered? This is a great piece of equipment for hiding computer windows from prying eyes, however, if one were setting up the TX in a darkened hallway, for instance, the blue LED light illuminating when someone passed by could easily draw unwanted attention to itself, which in turn could raise suspicions about its function and create more issues than it solves. Yes, the light does serve a useful function in its capacity to relay its status to the user, but wouldn't some kind of "stealth" mode during actual operation help to strengthen its usefulness.

dekuNukem, 2019-11-28:
Like you said, I do feel that the LED is useful for indicating what the TX is doing. Without it there's no way to see if it's triggering correctly, or if it's working at all.

I can indeed write you a custom firmware for you to disable the LED, but there would be no easy way to switch between normal and "stealth" mode. A simpler solution would be just cover it with a piece of black tape, faster and easier to undo.

geek-slt, 2019-11-29:
I would like to present to you one of my ideas. We are now using this to monitor pedestrians to trigger the computer to perform corresponding actions. We can imitate this model to make a privacy-protected hardware, that is, the sensor is placed in front of the computer to continuously monitor whether the person is at the table If you do n’t take action at the moment, if you ca n’t detect anyone, trigger the lock screen to prevent personal computer information from being leaked. Do you think this is feasible?

dekuNukem, 2019-12-11:
ah, so basically perform an action at the end of trigger instead at the beginning. I'll spend a few days working on it, probably having the RX press a different key combo.

monophthongal, 2019-12-18:
Seconding the request for a way to disable the blue LED...

The Daytripper device is fantastic, but hard to explain the presence of if someone asks - so it would be much more useful if it didn't attract attention to itself.

> Like you said, I do feel that the LED is useful for indicating what the TX is doing. Without it there's no way to see if it's triggering correctly, or if it's working at all.

For my use case, I'm OK with this since I can still test it by waving a hand in front of it and seeing if the windows are minimized as expected. The device has been 100% reliable for me so far, so that would be enough to tell me whether it's still on and functioning.

Also, the light on the TX could still flash a few times when first turned on - this would give you an initial indicator that everything is OK.

> A simpler solution would be just cover it with a piece of black tape, faster and easier to undo.

I tried this but there is still a lot of light leaking past the tape.

dekuNukem, 2019-12-18:
yeah that make sense. I have created a new TX firmware that disables the LED after initial calibration. The new firmware and the instructions can be found in the "USB Firmware Updates" section of this guide:

https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#usb-firmware-updates

monophthongal, 2020-01-06:
Had a chance to test the new firmware yesterday - works great! Thanks for adding it.

closed 2020-01-09 by dekuNukem