dekuNukem/daytripper #17: Having two close by

kcc1982, 2019-12-19:
I got my coworker one since I thought they were on different frequencies but it turns out we are locking each other's computers. Is there a way to have two of these in the same cube farm.

dekuNukem, 2019-12-19:
By default all daytrippers are on the same adress, so they will trigger each other if there are more than one.

I wrote you a custom firmware that uses a different address. Flash them on one of the TX/RX pairs, and they'll talk on a different address, allowing you to use two sets at the same time without interference.

Download the firmware here:

https://cdn.discordapp.com/attachments/514708977498587148/657235124378664980/alt_addr.zip

and refer to "USB firmware updates" section of [this guide ](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md#usb-firmware-updates) to see how to upload it.

kcc1982, 2019-12-29:
It worked. Thank you very much

https://cdn.discordapp.com/attachments/514708977498587148/657235124378664980/alt_addr.zip

closed 2020-01-02 by dekuNukem