dekuNukem/daytripper #18: broken rx

ringnex1, 2020-03-13:


varenc, 2020-03-27:
just got out my daytripper again and I'm seeing the same behavior.  I had things working fine a few months ago. What does it mean if all these values are zero?  TX also having issue it seems. 

One guess: Could this be because the static ram has been reset after being untouched for many months? Assuming the SRAM is kept alive/powered when these are being used regularly?  Could a firmware reprogram help solve this?

varenc, 2020-03-27:
actually nvm, my issue is almost certainly this issue: https://github.com/dekuNukem/daytripper/blob/master/mysterious_glitch.md

the 0 values on the rx are a red herring I suspect. My tx's battery is definitely touching the board

ringnex1, 2020-03-27:
Hey so I don't think my tx is the issue because I don't have the battery issue and also do you know if my new set might have got lost in shipping because it's not here yet

closed 2020-05-22 by dekuNukem