dekuNukem/daytripper #21: RX not receiving signal

nicolasnsuy, 2020-07-04:
Hi, i been not able to use my sensors for a while both TX and RX were working fine, now the RX turn on the blue light breathing and then go dim.

The TX turns on ok and the blue light turns on when you you trigger the sensor.

The pc is not making any action and if i press de debug button on RX doesn't show receiving any package from the TX.

I already try updating the firmware but they still doing the same.

Any idea of what can be happening?

Thank you!

dekuNukem, 2020-07-04:
Hi There! It sounds like a issue that was discovered in early-production run models. Please take a look at [this guide](https://github.com/dekuNukem/daytripper/blob/master/mysterious_glitch.md) and see if it helps. I do apologise if for the inconvience.

nicolasnsuy, 2020-07-05:
Thank you for your response.

I already follow all steps but with out any luck, the TX has a Air gap but still not working.

Is there anything else i can try?

Thanks!

dekuNukem, 2020-07-05:
Can you tell me your Tindie order number? I'll look into it.

nicolasnsuy, 2020-07-05:
Sure, Order Number: 174290.

Thank you!

dekuNukem, 2020-07-05:
Great! I'm happy to sort this one out for you. Please check your email.

closed 2020-07-05 by dekuNukem