dekuNukem/daytripper #22: Not recording keystrokes anymore

Saeijou, 2020-07-25:
Hello!

I really love your product, but now I encountered an issue.

I used it for a couple month with my Macbook Pro and in unison with BTT it worked amazingly!

Recently I went back to windows, and no dice.
It installed the "keyboard", but I can't get it to anything, no matter what mode I chose.

The lights are working as described and even if I trigger it, the receiver flashes in acknowledgement of receiving the message - yet windows does nothing.

I connected it to my macbook again, and also there it is not working anymore.
It is not recording the keystroke - even if I want to create a new script.

Did I break it by using it cross-OS? Any explanation?

dekuNukem, 2020-07-25:
Hi Saejiou:

Make sure the mode select switch is in the right place, if it is in `custom` it will press ALT+PAUSE which doesnt seem to do anything at all.

You can also try flashing the firmware again and see if it helps. You can find the instruction for both in [this guide](https://github.com/dekuNukem/daytripper/blob/master/advanced_usage.md).

Let me know how it goes!

Allen

Saeijou, 2020-07-25:
Thank you!

Interesting enough, it doesn't recognize the device it seems. I can't select "STM Device in DFU Mode"
The LED is off when I plug it in as described. (it is on if I plug it in normally)

dekuNukem, 2020-07-25:
This looks like something wrong with the physical USB port, the data isnt getting through. check that it is clean and undamaged. maybe try a different cable too.

Saeijou, 2020-07-26:
Splendid! It was the cable - i exchanged it for a shorter one when I switched computers.

Thank you so much!

closed 2020-07-26 by Saeijou